LemLib/LemLib #12: [FEATURE] Doxygen workflow

SizzinSeal, 2023-02-14:
### Requested Feature
When a new release has been pushed, the docs should be automatically updated to reflect new changes and version increments. This includes tutorials, doxygen documentation, and javadocs at the start of the source files.

### Benefits
- Easier maintenance

Exortions, 2023-02-14:
I'll work on this later today.

Had some issues with the Doxygen workflow last night, so hopefully I can get them ironed out today.
I'll also have a V5 brain to test new features tonight, so will be able to test other features.

SizzinSeal, 2023-02-14:
Doxygen macros can be used to expand things like the project version (e.x expanding LemLib@VER.zip to LemLib@0.4.2.zip), so only a single line has to be changed in the doxyfile

Exortions, 2023-02-14:
Re-opening to address other issues

SizzinSeal, 2023-02-23:
Doxygen does not run on push to master branch. It deploys the pages but does not update

commit bf197e05 (2023-02-14): add doxygen workflow (fix #12)

add doxygen workflow (fix #12)
PR #32 (2023-03-15): Documentation Workflow and Improvements
#### Summary
Fixes the doxygen workflow. It now builds the site, and deploys it on the `gh-pages` branch automatically. Updated documentation. Cleaned up root folder, greatly reduced size of master branch.

#### Motivation
Closes #12. Updated documentation to prevents confusion, and a clean root folder makes the project easy to navigate and understand.

#### Test Plan
- [X] Test build and deploy in fork

#### Additional Notes
[Preview](https://sizzinseal.github.io/LemLib/index.html)
closed 2023-03-15 by Exortions