LemLib/LemLib #16: [FEATURE] VSCode Extension

Exortions, 2023-02-17:
# Visual Studio Code Extension

This LemLib VSCode extension would provide support for many different functions:

 * Path creator (migrate from https://github.com/sizzinseal/path-gen)
 * Logging
 * Setup - setting up LemLib
 * [Maybe] Gui Creator - Create a teleop and autonomous through a GUI, without any programming

It would also include a few commands:

 * Install - installs LemLib using the PROS CLI
 * Upload path - uploads the generated path to the V5 Brain using a serial port
 * Save log - saves the log of the ran program to the output

SizzinSeal, 2023-03-03:
VSCode extension now has a repo: https://github.com/LemLib/Extension

SizzinSeal, 2023-05-02:
Closing this as this has graduated from an issue to an entire repo

closed 2023-05-02 by SizzinSeal