LemLib/LemLib #23: Need help setting up LemLib.

ItzDerock, 2023-03-03:
What makes a tracking wheel "reversed"?
For the left/right I'm assuming that moving forward should increase the encoder values, and for the middle, I'm not sure.

Here's a quick video that details the issue I'm running into and also the setup of my robot:
https://media.derock.dev/3PGWwB.mp4

Code: https://github.com/ItzDerock/95993A-spin-up/
You're probably interested in `src/config/config.cpp`

SizzinSeal, 2023-03-03:
I’ll take a look in 3-4 hours, busy with hw

SizzinSeal, 2023-03-03:
when it moves to the right, X should increase. From what I could see, your odom setup looks good

ItzDerock, 2023-03-03:
but when i turn the robot 90deg and line it back up to where it started originally, shouldn't the x,y still be 0 and theta is just 90?
and when the robot is facing right and I push it forward, the x decreases instead of increasing. 

https://derock.media/xxTjiR.mp4

SizzinSeal, 2023-03-03:
Oh my bad, i didnt realise the clip had sound. So the x and y changing when turning on the spot indicates the tracking wheels are improperly configured, as you suspect. In addition, the robot should be facing 90 degrees, not -90 when turning right. These indicate 2 separate issues:

**Issue 1 - Vertical tracking wheels**
Looking at your code, your left and right tracking wheels seem to be set up correcly. The left tracking wheel has a negative distance value, and the right tracking wheel has a positive distance value, which corresponds to the robot in the video. However, the robot thinking it has turned `-90` degrees instead of `90` degrees shows that your left and right tracking wheels have their ports swapped.

**Issue 2 - Horizontal tracking wheel**
Since you horizontal tracking wheel is behind the tracking center, it needs to have a negative distance value. This is most likely the cause of the x and y changing on the spot.

Hope this helps