LemLib/LemLib #28: [BUG] Logger System Unused

SizzinSeal, 2023-03-12:
#### Observed Behavior
The debugger system is unused. 

#### Expected Behavior
Making mistakes such as trying to move the chassis prior to calibrating should be logged appropriately.

#### How to Reproduce
1. Configure chassis
2. Move chassis without calibrating it

#### Project Info
Pros Kernel: 3.8.0
LemLib version: 0.4.5
Additional Libraries: None

#### Errors / Screenshots / Code
None

#### Additional Notes
Some examples of when the logger should be used:

1. Turning to face a point very close to the robot [Warning]
2. V5 device command failed to run successfully [Error]
3. Chassis sensor calibration failed [Error]
4. Odom configuration error (e.g 2 tracking wheels in same position) [Error]
5. Failed to read SD card [Error]
6. Improperly formatted file [Error]
7. Same motors on left and right side drivetrain [Error]

and many more situations that should be addressed in a PR

Exortions, 2023-05-02:
Merge the PR, it's done iirc

SizzinSeal, 2024-04-19:
Implemented in master branch

closed 2024-04-19 by SizzinSeal