LemLib/LemLib #36: [FEATURE] PTO support

SizzinSeal, 2023-05-02:
#### Requested Feature
Power Take Off support in the Chassis class

 - Should be able to dynamically switch motors from other subsystems to the chassis
 - The extra motors should be included in odometry calculations (if using motor encoder odometry)

#### Current Implementation
It is not possible to do this.

#### Motivation
PTOs are becoming increasingly common, and users should be able to make use of them when using LemLib

Exortions, 2023-05-30:
Assigned to Rocky :)

SizzinSeal, 2023-09-26:
Waiting on PROS 4 release. It will add the ability to add and remove motors from motor groups, making PTO implementation very simple.

ItzSt0rmz, 2023-11-26:
@Rocky14683 PROS 4 is out and used in the refactor right?

Exortions, 2023-11-29:
PROS 4 is not used in the refactor

SizzinSeal, 2023-11-29:
Pros 4 is used in the refactor

Exortions, 2023-12-05:
oh my bad, i thought it was still on 3

ItzSt0rmz, 2023-12-08:
I should be able to do this during/after exams next week.

SizzinSeal, 2025-01-04:
Implemented in master

closed 2025-01-04 by SizzinSeal