LemLib/LemLib #46: [Gain Scheduling]

Nightsoot, 2023-06-11:
#### Requested Feature
 The ability to put in control cases for PID constant(s) (I.e. kP = 2, kD = 3 at 90 degrees and kP = 2, kD = 4 at 160 degrees) and interpolate between the constant(s).

 - a Once the movement starts the constants should be calculated, they don't need to change during the movement.

#### Current Implementation
 There currently is no implementation.

#### Motivation
 Due to the non-linear nature of the robot's movement, one set of PID constants is not as effective in one case to the other, so it would be nice to have a way of automatically making adjustments to the constants to increase consistent performance.

#### (Optional) Possible Implementations and Alternatives
 Having an alternative constructor for FAPID, where you can insert control cases as list of lists or other method (I.e. {kF = 1, kA = 0, kP = 3, kI = 1, kD = 4, case = 90}) and from lowest case to highest case, interpolate linearly or with a spline the constants

SizzinSeal, 2023-06-12:
This is planned for the next major version (v0.5.0). Unfortunately this will sit in the backlog for a bit as there are some huge pull requests needing to be merged

SizzinSeal, 2023-06-12:
Additionally, this feature does complicate the PID class quite a bit. An important goal of the library is to remain simple, so I'll have to think of a clean way to do this. Let me know if you have any ideas

Nightsoot, 2023-06-12:
Fair enough, I understand that simplicity is an important concern, so could you specify what you're trying to avoid other than just general design issues, I'd be willing to help out I just want to know if there is a certain pattern to how you want the PID class structured. Thanks, also we could move this conversation to Discord if that's easier.

SizzinSeal, 2023-06-12:
Discord would be easier, as convo is about designing the solution, something GitHub issues aren't super great at