LemLib/LemLib #66: [FEATURE] Correctly handle motor oversaturation.

Tropix126, 2023-09-20:
#### Requested Feature
When combining two PID outputs together (such as in the case of point-to-point movement, which runs an angular and linear controller), the combined outputs may end up being more than the maximum motor voltage (128, or 12v). This is less than ideal, because the ratio between the left and right wheel speed will not be preserved, and the robot may turn more or less than expected (depending if the left or right voltage is outside (-128, 128)).

#### Current Implementation
https://github.com/LemLib/LemLib/blob/master/src/lemlib/chassis/chassis.cpp#L311-L317

#### Motivation
See above paragraph. Point-to-point movements currently will not execute exactly as expected, and may reach the target slower.

#### (Optional) Possible Implementations and Alternatives
Normalize the left and right wheel volatages. If one voltage is over the saturated value (128), then decrease both voltages to preserve the ratio between the two:
[image]

Example implementation ripped from some old code:
```cpp
std::pair<double, double> normalize__motor_power(double left_power, double left_power, double max_power) {
	double largest_speed = std::max(std::abs(left_power), std::abs(power_speed)) / max_speed;

	if (largest_speed > 1.0) {
		left_speed /= largest_speed;
		right_speed /= largest_speed;
	}

	return { left_speed, right_speed };
}
```
I think this actual algorithm originates from wpilibc, but it's been used in other places as well (such as okapi).

SizzinSeal, 2023-09-20:
You will notice above that code snippet you posted there is a variable called overturn. It prioritizes turning over moving, to help with a drawback of this specific motion alg. I do really appreciate how much detail you went into. I'll leave this open til you respond :)

Tropix126, 2023-09-20:
Oh whoops, I totally missed that 😅. Closing as resolved.

closed 2023-09-20 by Tropix126