LemLib/LemLib #68: Logger not correctly logging Integral term in FAPID [BUG]

Nightsoot, 2023-10-01:
#### Expected Behavior
When logging PID it should show P term, I term, D term as they should be calculated

#### Observed Behavior
The logger shows an I term regardless of whether or not kI = 0

#### Steps to Reproduce
Log PID with a kD not equal to 0

#### Environment

 - LemLib Version: movement_beta 
 - PROS Kernel Version: 3.8.0
 - PROS CLI Version: 3.4.1
 - Additional Templates: None

#### Additional Information
Found the issue in pid.cpp under lemlib::FAPID::update(), where the line to log is using kD for the Integral term instead of kI

`if (log) { Logger::logPid(name, output, error, kP * error, kD * totalError, kD * deltaError); }`

SizzinSeal, 2023-10-01:
This is working correctly. Its not showing the integral gain, which is 0, but its showing the integral value. Even though the integral value has no effect on the output, its still being kept track of

Nightsoot, 2023-10-01:
Why is the totalError being multiplied by kD then?

SizzinSeal, 2023-10-01:
Sorry my brain left my body there. You are correct. My apologies. This will be fixed within the next update, waiting on the logger rewrite pr #65

SizzinSeal, 2024-04-10:
pid logger has been removed

closed 2024-04-10 by SizzinSeal