LemLib/LemLib #83: [FEATURE] Motion chaining

SizzinSeal, 2023-10-27:
#### Requested Feature
Motion chaining. Transition from on motion to the other smoothly without stopping

 - Robot settles before moving on to next point
 - Unnecessary deceleration and acceleration
 - Wastes time

#### Current Implementation
Set timeout too a value too small so the robot does not finish the path and translate the target.

#### Motivation
Saves time in autos

#### (Optional) Possible Implementations and Alternatives
Chaining will look different for every motion algorithm:

###### Boomerang:
Find the imaginary line perpendicular to the heading of the robot. This line intersects the target point.
If the robot can carrot point was on the same side of the line last cycle, but on different the current cycle, exit.
Motion chaining mode enabled whenever a parameter called minimumSpeed is set to a non-zero value

###### Turning
Exit when the robot crosses the target heading. Does this whenever a parameter called minimumSpeed is set to a non-zero value

###### Pure Pursuit
Exit whenever the closest point is the last point or the ghost point

Exortions, 2024-02-06:
is this ever gonna happen? seems like a cool feature

SizzinSeal, 2024-04-10:
preset in latest pre-release