LemLib/LemLib #84: [FEATURE] Boomerang controller improvements

SizzinSeal, 2023-10-27:
#### Requested Feature
Turn smoothness parameter and secondary carrot lead gain.

 - Turn smoothness parameter. Value from 0 to 1
 - gLead. A second gain for the carrot point.

#### Current Implementation
Currently has dLead, chasePower, and maxSpeed tuneable values, as well as PID gains

#### Motivation
Greater control over the boomerang controller and the curve it makes.

#### (Optional) Possible Implementations and Alternatives

###### Turn Smoothness:
```c++
float overturn = fabs(angularPower) + fabs(linearPower) - maxSpeed - maxSpeed * smoothness;
if (overturn > 0) linearPower -= linearPower > 0 ? overturn : -overturn;
prevLinearPower = linearPower;
```

###### gLead:
```c++
currentCarrot = initialCarrot + (currentCarrot - initialCarrot) * (1 - gLead)
```

Exortions, 2023-11-20:
Is this getting fixed in the refactor?

SizzinSeal, 2025-01-04:
Won't implement. LemLib will only provide improved algorithms through written tutorials in future

closed 2025-01-04 by SizzinSeal