LemLib/LemLib #102: Custom moveVelocity Function

ItzSt0rmz, 2023-11-22:
#### Requested Feature
Create a custom moveVelocity function to replace the flawed VEX one

#### Current Implementation
Currently LemLib just uses voltage for moving motors

#### Motivation
Easier and more intuitive for movement algorithms, chaining off of issue #4 

#### (Optional) Possible Implementations and Alternatives
Could do something uses move voltage and a PID loop, with get_velocity as the input for the loop
(In conjunction with above) Could also create some sort of function to predict a starting point for the voltage to minimized time to reach target

ItzSt0rmz, 2023-11-22:
Could also look at other controllers like a TBH

SizzinSeal, 2023-11-22:
yea this will definitely be a thing once velocity controllers are implemented

ItzSt0rmz, 2023-11-26:
Update: After discussion on the discord, it looks like the best proposed solution is a Velocity Acceleration Proportional loop.

SizzinSeal, 2025-01-04:
Won't implement. LemLib will only provide improved algorithms through written tutorials in future

closed 2025-01-04 by SizzinSeal