LemLib/LemLib #169: Swing turn radius parameter

Pointy-Carrot, 2024-04-16:
#### Requested Feature

The swingToPoint() and swingToHeading() functions should have radius parameter to change the swing arc.

 - .radius will be a new parameter in the params struct and have an default value of 0
 - increasing .radius will increase the swing radius in inches
 - a value greater than 0 of .radius will necessitate the use of the locked side of the drivetrain to achieve the specified swing radius

#### Current Implementation

Currently you can do janky stuff with boomerang to achieve a similar result.
#### Motivation

Using moveToPose to simulate a swing turn with a larger radius is janky and not optimal. A .radius parameter would increase the versatility of the swing turn functions while not noticeably increasing the complexity of the function call and usage.
#### (Optional) Possible Implementations and Alternatives

Rocky14683, 2024-04-16:
Swing doesn't use radius. It only use angle error and lock one side of the wheels.

Pointy-Carrot, 2024-04-17:
> Swing doesn't use radius. It only use angle error and lock one side of the wheels.

Yes, this is the currently implementation of the swing functions. However, it would be beneficial to add a new parameter called ".radius" in the params struct for the functions to increase their versatility. A sample use case would be if the robot is on one side of the gray short barrier, you could use a swing function with increased radius to swing all the way around the barrier.

PA055, 2024-05-15:
I'm going to try my hand at this enhancement, and since it should be mostly math and not too much code, it should be easy to transfer to 0.6

SizzinSeal, 2025-01-04:
Won't implement. LemLib will only provide improved algorithms through written tutorials in future

closed 2025-01-04 by SizzinSeal