LemLib/LemLib #198: [FEATURE]

Pointy-Carrot, 2024-05-07:
#### Requested Feature
Users should be able to easily mirror a path and/or movement functions for different alliance sides of the field.

 - User should be able to call chassis.setDefaultAlliance() in chassis config to choose what alliance perspective they normally code their autons on.
 - At the beginning of the autonomous code, the user should be able to call chassis.setAlliance() to select the alliance they are on.
 - If the user set their default alliance as red and sets the current alliance to blue at the beginning of auton, the corresponding red side auton will run but with the x position signs and theta signs flipped so that the auton is effectively mirrored.

#### Current Implementation

User has to make a completely new path file which is a mirror of the original path.
#### Motivation
Manually mirroring autonomous routines is annoying and prone to mistakes. This feature would insure that no more mistakes are made when mirroring paths and cuts down the time spent on creating autons.

#### (Optional) Possible Implementations and Alternatives

ABUCKY0, 2024-05-07:
Really should give a better title

SizzinSeal, 2024-05-08:
This will be introduced in v0.6. This requires significant modification to the tracking algorithm and it would be difficult to implement considering the complexity of the tracking algorithm in v0.5.

We'll do our best to get v0.6 stable before the end of the summer holidays