LemLib/LemLib #208: [FEATURE] ♻️ Refactor Chassis Class

Aang099, 2024-05-18:
#### Requested Feature
The Chassis class should be god-like in structure. However, logic should be contained in separate, independent classes.

With this refactor proposal, chassis class will retain functions such as moveToPoint, but instead of having the motion algorithm logic containted within that function, it would, for example, use the parameters to instantiate a new AdaptivePIDSeeking class and pass it to a Motion pointer. Chassis::moveToPoint will essentially be a wrapper

The chassis class will also be able to be constructed with dependency injection as an alternative to the configuration classes. This allows for great extensibility by advanced users and makes external LemLib extensions possible 

#### Motivation
Users interact with the chassis class 95% of the time in all previous versions of LemLib. It's proven to be simple user-side. However, it means the logic on the implementation side is very complex. This aims to compromise between the two, hence the term "god-like"

montymi, 2025-03-12:
Hello, I noticed that this feature request [#208](https://github.com/orgs/LemLib/projects/5?pane=issue&itemId=63389598&issue=LemLib%7CLemLib%7C208) is in progress and I would love to assist with its development. I also see that the linked PR was merged to master. Please let me know how I can help and if there are specific tasks or areas you would like me to focus on.

SizzinSeal, 2025-03-13:
@montymi I'll open some issues tomorrow with specifications. I would do so tonight but I'm running on 4 hours of sleep and have made 150 commits in 3 days. Sorry for the late response