LemLib/LemLib #228: [FEATURE] Dynamic Lookahead Distance

SizzinSeal, 2024-08-22:
#### Requested Feature

Dynamic lookahead distance

 - Lookahead distance scales automatically to better follow the path
 - Lookahead distance changes while the path is being followed

#### Current Implementation

Lookahead is static and manually input by the user

#### Motivation

Pure Pursuit is known to deviate from the path, especially around corners, before converging with it again. A dynamic lookahead distance can reduce this deviation around corners while maintaining speed around straight sections.

#### Possible Implementation

__**let**__:
 - $l$ represent the lookahead distance
 - $l_{min}$ represent the minimum lookahead distance
 - $l_{max}$ represent the maximum lookahead distance
 - ${v_t}$ represent the target velocity

 $l = {clamp}(v_t \cdot k_l ,  l_{min} ,  l_{max})$
 
 With this implementation, the lookahead increases at high speeds and decreases at lower speeds, but will not exceed the specified limits. This could be implemented either as a calculation to be performed at runtime

Vlarkus, 2024-08-22:
Is the suggestion to modify how interpolation of waypoints on a path happens by also considering the speed of the robot at that line segment? For example, driving slower on a specific segment would decrease LD and therefore add more waypoints?

SizzinSeal, 2024-08-22:
This issue does not suggest changing the interpolation of the waypoints. The algorithm will just find the lookahead point on the furthest line segment, according to the current implementation

SGD2718, 2024-08-22:
At higher speeds, the look-ahead distance $\ell$ should be greater than it would be at a lower speed so that sharp turns and such may anticipated earlier. This motivates us to make $\ell$ proportional to the target velocity $v$. The target velocity is used due to its stable nature (as opposed to measured velocity). Let target velocity be bounded by $v_{min} \leq v \leq v_{max}$ for all possible paths. Let $\ell_{max}$ be the maximum look-ahead distance. We may set $$\ell = \frac{v}{v_{max}}\ell_{max}$$, which will ensure that $\ell = \ell_{max}$ when $v =v_{max}$ and $\ell = \frac{v_{min}}{v_{max}}\ell_{max} > 0$, when $v = v_{min}$.

I am unsure of whether the look-ahead point should regress if $\ell$ shrinks enough so that distance from the robot's current position $R$ and the old look-ahead point $L_{prev}$ is greater than $\ell$. The dynamic look-ahead distance should be tested with and without this regression. 

**Possible implementation of look-ahead point regression:**
Let:
- $W_i$ denote the $i^{\text{th}}$ waypoint from the beginning of the path.
- $i_{0}$ be such that $W_{i_0}$ is the closest waypoint to the previous look-ahead point.
- $\operatorname{distance}(R, W_{i})$ denote the distance from the robot to $W_i$
- $\ell$ be the new look-ahead distance
- $E(i_k) = \text{abs}\left(\operatorname{distance}\left(R, W_{i_k}\right) - \ell\right)$ be distance error.
If $E(i_{0}-1) > E(i_0)$, then let $i_{1} = i_{0} - 1$, and more generally, let $i_{k+1} = i_{k}-1$ until $E(i_n) < E(i_{n+1})$. Otherwise, let $i_{k+1} = i_k + 1$ until the same condition occurs. If $\operatorname{distance}(R, W_{i_n}) \leq \ell$ and $\operatorname{distance}(R, W_{i_{n+1}}) \geq \ell$ or vise versa, interpolate between $W_{i_{n}}$ and $W_{i_{n+1}}$ to find the look-ahead point. Otherwise, it will lie between $W_{i_{n-1}}$ and $W_{i_{n}}$.

SizzinSeal, 2025-01-04:
Won't implement. LemLib will only provide improved algorithms through written tutorials in future

closed 2025-01-04 by SizzinSeal