LemLib/LemLib #265: New motion algorithm

tmassoni13, 2025-02-02:
#### Requested Feature
 Relative movements that aren’t actually relative but are coded like relative movements by the user.

 - a: moveToPoint revamped to be used like a relative motion that generates a target position for the user based on their last target position and heading.
 - b: would take arguments similar to JAR’s move_distance function such as distance in inches and final heading.
 - c: goes off of the users most recent target position to generate the new point so if the auton gets off target the next movement will be unaffected. 

#### Current Implementation
 
there is no implementation as of right now. And I don’t know how difficult or feasible it would be.

#### Motivation

It would make lemlib much easier to code with for current users and would be an easier gateway to new users.

#### (Optional) Possible Implementations and Alternatives
 N/A

meisZWFLZ, 2025-02-02:
Related to: https://github.com/LemLib/LemLib/pull/134

tmassoni13, 2025-02-02:
Addition: maybe there could also be a way to choose how far into the movement the robot starts to head to the final heading for instance drive forward 12 inches but at 6 inches start turning towards a final heading of 45.