LemLib/LemLib #268: [BUG] Incorrect pose coordinates after turning

Hexa-Force, 2025-02-04:
#### Expected Behavior

Turning in place has no effect on x and y position.

#### Observed Behavior

Making a turn with chassis.turnToPoint or chassis.turnToHeading results in an extreme change in coordinates when printing the robot's pose before and after the turn using chassis.getPose(). For example, our robot had a pose of "X: 60.024670, Y: 0.478130, Theta: 349.325897" before a turn. After turning to a heading of 270, the pose was "X: 50.802494, Y: 1.038919, Theta: 271.200562." The robot made the correct physical movements for the turn, but everything after was thrown off because the X coordinate after the turn was off by 10 inches. This is just a single example. We run into this issue with almost every turn. 

#### Steps to Reproduce

Run `chassis.turnToHeading(270, 3000, {.maxSpeed = 80}, false);` and print the pose before and after. 

#### Environment

 - LemLib Version: 0.5.5 
 - PROS Kernel Version: 4.1.1 
 - PROS CLI Version: 3.5.4 
 - Additional Templates: 

#### Additional Information

toxicdefender404, 2025-02-04:
Not a bug.
You need to adjust the offsets for your tracking wheels (or the trackwidth if you have no tracking wheels) or the robot's tracking center will not necessarily align with the robot's turning center, causing the coordinates of the tracking center to change when the robot rotates

slipperking, 2025-02-05:
Your tracking center likely wont be the dead center of the robot due to balance issues. Anyways, the robot left right front back should be determined from the back view.

If your "front" is the mogo side, then using the imu hole as the tracking center, the horizontal offset should be positive, and the vertical wheel offset is negative.

If the front is the intake side, then the horizontal has negative offset and the vertical wheel will have positive.
So one of the two wheels should have a positive offset and the other negative with your current offset

Also lemlib has a bug, use "2" instead of "lemlib::Omniwheel::NEW_2". When you test note that a tile is slightly under 24 inches

Hexa-Force, 2025-02-05:
It turns out we measured our offsets incorrectly the first time. We ended up making the vertical offset positive like you said, and that definitely improved the accuracy of the coordinates, but we're still having the same issue. We also switched the omniwheel diameter to 2", and that improved the accuracy of lateral movements, but that didn't have any effect on the issue of coordinates changing after turning. We confirmed that the inertial sensor is mounted in the center of the robot. Even if it's not exactly at the center of mass, it is really close to that. Here's the most recent test with a turn to point that should have been 90 degrees. 

`chassis.turnToPoint(-48, 24, 1000, {.forwards = false, .maxSpeed = 70}, false);`

Before turning to point: X: -47.850502, Y: -0.207907, Theta: 89.991684 //This is correct
After turn X: -54.704185, Y: -4.887868, Theta: 190.358932