LemLib/LemLib #269: [FEATURE] MovementWise Tracking Offset

NoahK216, 2025-02-08:
#### Requested Feature
Allow a tracking offset to be specified for each motion. This would allow for motions to move until a that offset from the robot's tracking center is at a position.

v0.5 eg:
```
// Move until the point 12 inches in front of and 12 inches to the right of your robot is at the origin
chassis.moveToPoint(0,0,1000,{.trackingOffset={12,12,0}});
```

#### Current Implementation
None

#### Motivation
Useful in situations where the location of a subsystem is important. A primary example being a moveToPoint that easily ends with a goal rush mech on a goal.

NoahK216, 2025-02-08:
I see that this is very similiar to [issue 232](https://github.com/LemLib/LemLib/issues/232). I think it makes more sense to have it a per movement basis but this can be closed if it's annoying.

NoahK216, 2025-02-11:
@meisZWFLZ Suggested:
`chassis.withOffset({12,12}).moveToPoint({0,0});`
"Where withOffset() returns a transformed chassis object that to the user behaves the same as the normal chassis object"

toxicdefender404, 2025-02-20:
can't you currently already do this `` setPose({getPose().x + 12, getPose().y + 12, getPose().theta})``