LemLib/LemLib #273: [BUG] v0.5.4 does not correctly handle motion tasks being killed by field control

aadishv, 2025-06-16:
take this PROS code in main.cpp:
```cpp
lemlib::Chassis chassis(...);
pros::Motor intake(...); // for signalling
void initialize() { chassis.calibrate(); }
void autonomous() {
  chassis.setPose({0,0,0});
  chassis.moveToPoint(0, 0, 1000, {}, false);
  intake.move(127);
}
```
Configure chassis ports, etc., correctly, and tip the drivetrain on its side (so the wheels are not contacting the ground).
*Test 1*
* Calibrate chassis in disabled.
* Enable autonomous mode. The motors move for 1 second, then time out. Intake then starts moving.
* Once the intake starts moving, disable and then reenable autonomous. The motors will again move then time out, and then the intake will start moving.
All of this behavior is expected.
*Test 2*
* Calibrate chassis in disabled.
* Enable autonomous mode. This time, disable it before it times out (i.e., disable it while the drivetrain is still moving and intake has not started moving.
* Reenable autonomous mode. 
Expected behavior: the motors move for 1 second, time out, then intake starts moving.
Reality: the motors do not move, and the motion does not time out. Nothing moves, even after 1 second.
*TL;DR: LemLib does not properly handle disabling in the middle of a motion.*
I think this is really two separate problems:
1. LemLib does not properly clear the task when the robot is disabled, so some edge case occurs preventing the new task from getting set.
2. Because the new task didn't get set, it doesn't time out. This means that no further motions can complete, and in the second autonomous period, the robot hangs on the motion.
(by task, I do not mean an RTOS task - I mean the currently executing motion)

meisZWFLZ, 2025-06-16:
## Potential diagnosis:
Pros deletes the autonomous task when switching to disable ([pros code](https://github.com/purduesigbots/pros/blob/40c9131b33f455df74218a1dcdf7039c8d69496b/src/system/system_daemon.c#L109C1-L116C1)).
That then means the chassis never gets the chance to free the mutex, and afaik the mutex will not be freed when the auton task is deleted. ([freertos forum post](https://www.freertos.org/FreeRTOS_Support_Forum_Archive/May_2011/freertos_Delete_task_holding_a_mutex._Is_it_released_4546788.html))

Im not entirely sure of this though. It should be easy to test by modifying the header file to expose the mutex, and print whether the mutex is locked at the start of the motion.

## Possible Solution
Spawn all motions inside a task, so that they dont get deleted and can free their mutexes correctly.

aadishv, 2025-06-16:
I don't currently have access to a brain/drivetrain to work with, so for now I'll assume this is the issue.

Two main changes to be made (to every chassis motion function):
* Regardless of whether you called with async = true or not, LemLib **always** spawns a new task to run the PID/etc. loop. The only difference is that if async = false, there is a call to `waitUntilDone` afterwards to block. This way, motions always run in external tasks, which are not canceled like competition tasks are and thus can safely time out.
* Have motion loops detect if they are in disabled, and if so, exit cleanly. This avoids the issue where a motion with timeout 10,000 will act like a ghost and start moving again after the drivetrain is reenabled (which would also block other motion tasks from running).

Implications:
* Slightly more overhead as all motions now call a task. 
* This will only fix undefined behavior for teams that have LemLib motions which lock the mutex in competition tasks such that these motions run both before and after the drivetrain is disabled. This most notably applies to VEX AI teams.

aadishv, 2025-06-17:
https://github.com/LemLib/LemLib/pull/274 should resolve this